prusa3d/PrusaSlicer #12376: mk3.5 mmu3 single fillament mode

cozmo14047, 2024-03-03:
Hi there, for the mmu3 on the mk3s+ there is a mmu single fillament mode where it auto removes the purge tower and asks what port of the mmu to use for that print, as for the mk3.5 there is only a multicolor version so you have to chose the fillament on the slicer or go on to the screen and reset the port on the printer. please add the mk3 option for single filament. many thanks

[image]

rtyr, 2024-03-03:
That is intended. Use standard MK3.5 printer profile (non MMU) for "single mode" prints (the printer also lets you select the extruder).

closed 2024-03-03 by rtyr

cozmo14047, 2024-03-03:
> That is intended. Use standard MK3.5 printer profile (non MMU) for "single mode" prints (the printer will then also let you select the extruder).

Ok thank you, i will let the others on the forum know too