prusa3d/PrusaSlicer #12378: WARNING: Bed temperatures for the used filaments differ significantly despite the fact that I use the same type of filament

feri2303, 2024-03-03:
### Description of the bug

Hi, in the latest versions I get this error message, despite the fact that I use the same type of material when printing. I only have PETG stored in the MMU2 tray.

**WARNING: Bed temperatures for the used filaments differ significantly despite the fact that I use the same type of filament**

[image]

Thank you

### Project file & How to reproduce

[Warning.zip](https://github.com/prusa3d/PrusaSlicer/files/14472325/Warning.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Latest 2.7.2

### Operating system

Windows 10 22H2

### Printer model

Prusa MK3S+ MMU2

Jan-Soustruznik, 2024-04-11:
Hi @feri2303, the current implementation behaves as intended. There isn't a check in place to verify whether a material has been used or not. The warning is consistently shown whenever a material with a different temperature is selected from the list in the side panel.

Jan-Soustruznik, 2024-06-24:
Improved in https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-alpha5 . Closed as solved.

closed 2024-06-24 by Jan-Soustruznik