prusa3d/PrusaSlicer #12381: Slicer silently generates a toolpath outside of print area, causing Y-axis crash during print

orosam, 2024-03-03:
### Description of the bug

When slicing the attached 241mm*180mm model with 0.2mm layer height, at one point an out-of-area G2 move is generated, which causes an Y-axis crash.

The model fits the print bed comfortably, and there is no error after slicing about any out-of-area toolpaths.

### Project file & How to reproduce

Steps to reproduce:

1. Load the attached model, `plate.3mf`.
2. Slice it with the configuration in the model (all of them are system presets from config version 1.12.1):
   * Print settings: any 0.2mm print profile (could not reproduce with other layer heights).
   * Filament: Prusament PETG (or Generic PETG, or Prusa PETG; could not reproduce with e.g. PLA profile).
   * Printer: Original Prusa MK4 Input Shaper 0.4 nozzle.
3. Note that there are no warnings.
4. Export the gcode.
5. Look for the offending line:

   `G2 X106.953 Y-23.005 I-54.947 E-.19528`

This line will cause an Y-axis crash.
[plate.zip](https://github.com/prusa3d/PrusaSlicer/files/14474000/plate.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2+win64, config_version 1.12.1

### Operating system

Windows 10

### Printer model

Original Prusa MK4 5.1.2+13478

rtyr, 2024-03-04:
Reproduced, thanks.

hejllukas, 2024-03-04:
SPE-2184

FidelCapo, 2024-03-21:
Fixed in [PrusaSlicer 2.7.3-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1). Closing.

closed 2024-03-21 by FidelCapo