prusa3d/PrusaSlicer #12385: 2.7.2 Error encountered didn't before this version.

hondaman74, 2024-03-03:
### Description of the bug

I upgraded to the 2.7.2 this morning when I got the notification update in the app. and now I am getting this error trying to slice a model that I have used before many times with no issues but now getting this error

Voronoi cell doesn't contain source point!

I am probably going to go back to previous version but it's with any printer and any profile that I select.  And it's only with one of the models in the image.  But why now and I getting this error

### Project file & How to reproduce

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/14474356/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.72

### Operating system

Mac Sonoma 14.3.1

### Printer model

Ender 3 V2, Anycubic Kobra 2 Max

rtyr, 2024-03-04:
Please provide your 3MF project file (the zip contains only STL).

TobiasLillig, 2024-04-17:
Hi, same error when slicing 20 multicolor models. MMU3

FidelCapo, 2024-04-30:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2024-04-30 by FidelCapo