prusa3d/PrusaSlicer #12387: 2.7.2 Internal Error: Access violation

techis, 2024-03-03:
### Description of the bug

When trying to slice a model, I got an internal error. I have 42 instances of the same model on the build plate. 

4 walls, 5 top and bottom layers, 40% infil. 

I have attached the error from the dialog box as well as a screen cap of the dialog box. Dismissing the dialog box causes prusa slicer to crash. 

16:09:03: Internal error: Access violation - no RTTI data!
16:09:03: Internal error: Access violation - no RTTI data!

[image]

### Project file & How to reproduce

[LED_Bar_Clip_Misumi_version2.zip](https://github.com/prusa3d/PrusaSlicer/files/14474805/LED_Bar_Clip_Misumi_version2.zip)

In the Zip file, I placed an 3mf file before clicking the slice button as well as the model I am slicing. 

The first time I sliced the model it failed and caused prusa slicer to crash with the error in the screenshot above. Upon reopening and repeating the process, the error does not seem to occur. I am going to submit this anyway in case it is an intermittent error. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11

### Printer model

Mk 3S+ w/ MMU3 Single

rtyr, 2024-03-04:
I was not able to reproduce the issue.

Trecares, 2024-04-06:
I had the same issue when slicing a different model. Initially I was using 2.7.3+Win64. Upon launching again I updated to 2.7.4+Win64 and tried again and the slicer crashed with the same error again. I remade it and saved the 3mf and it didn't crash again after trying to slice several times.

Version of PrusaSlicer
2.7.3 + 2.7.4

Operating system
Windows 10

Printer model
MK3S

Zip file has the 3MF and STL file. Log and screenshot attached below.
[corner-connector.zip](https://github.com/prusa3d/PrusaSlicer/files/14894333/corner-connector.zip)
[log.txt](https://github.com/prusa3d/PrusaSlicer/files/14894336/log.txt)
[image]

gath-cmd, 2026-09-28:
I have found that running prusaslicer in efficiency mode through the Windows Task Manager solves this issue.