prusa3d/PrusaSlicer #12388: Layer change G-code not being inserted anymore before the first layer

nmaggioni, 2024-03-03:
### Description of the bug

As of 2.7.2, the custom before/after layer change G-code snippets are not inserted anymore at the start of the first layer.
I did not find mentions of this new behavior in the release changelogs, is it intended?

---

v2.7.0 produced this output (initial embedded image preview and config dump truncated for brevity; was also affected by #11843):
```gcode
; printing object issue.stl id:0 copy 0
; stop printing object issue.stl id:0 copy 0

;TYPE:Custom
M104 S0  ; prevent additional set&wait commands
M140 S0  ; prevent additional set&wait commands
_LAYER TOTAL=520  ; set total layer count
PRINT_START BED_TEMP=115 EXTRUDER_TEMP=245 MATERIAL=ABS
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
;0.2
G1 Z.2 F18000
;AFTER_LAYER_CHANGE
;0.2
_LAYER CURRENT=1
G1 E-.75 F3000
G1 X89.589 Y83.872 F18000
G1 Z.2
G1 E.75 F1800
M204 S1000
```

And v2.7.2 is missing the custom layer change macros:

```gcode
; printing object issue.stl id:0 copy 0
; stop printing object issue.stl id:0 copy 0

;TYPE:Custom
M104 S0  ; prevent additional set&wait commands
M140 S0  ; prevent additional set&wait commands
_LAYER TOTAL=520  ; set total layer count
PRINT_START BED_TEMP=115 EXTRUDER_TEMP=245 MATERIAL=ABS
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 E-.75 F3000
G1 Z.6 F18000
G1 X89.589 Y83.872
G1 Z.2
G1 E.75 F1800
M204 S1000
```

The diff highlights that, in addition to the initial Z-height having been fixed and the feedrate not being repeated for the first XY move, the {BEFORE,AFTER}_LAYER_CHANGE macros are skipped:
```diff
14,19d13
< ;BEFORE_LAYER_CHANGE
< ;0.2
< G1 Z.2 F18000
< ;AFTER_LAYER_CHANGE
< ;0.2
< _LAYER CURRENT=1
21c15,16
< G1 X89.589 Y83.872 F18000
---
> G1 Z.6 F18000
> G1 X89.589 Y83.872
```

---

For reference, in addition to the attached project file, here are my G-code snippets. They are used by Klipper to handle actions on specific layers and by postprocessing scripts to determine other metrics. Most importantly, `_LAYER CURRENT=1` is currently not included anymore in the final output and that breaks some counts in both processes.

+ Start:
```gcode
M104 S0  ; prevent additional set&wait commands
M140 S0  ; prevent additional set&wait commands
_LAYER TOTAL=[total_layer_count]  ; set total layer count
PRINT_START BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP={first_layer_temperature[initial_extruder]} MATERIAL=[filament_type]
```

+ Before layer change:
```gcode
;BEFORE_LAYER_CHANGE
;[layer_z]
```

+ After layer change
```gcode
;AFTER_LAYER_CHANGE
;[layer_z]
_LAYER CURRENT={layer_num+1}
```

### Project file & How to reproduce

[Config export](https://github.com/prusa3d/PrusaSlicer/files/14474987/Untitled.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Linux (Manjaro)

### Printer model

Voron 2.4

SachCZ, 2024-03-04:
Hello @nmaggioni, it is indeed **intended** as there is not really a layer change happening at the beginning of the print.

When it comes to your concern with **specific layer actions/post processing scripts**: The reasoning on our side was that you should be able to **substitute** the lost functionality by using the custom **Start G-code**, which is also executed before the first layer. For example in your case setting the last line of `Start G-code` to `_LAYER CURRENT=1`. Is there something we are missing that prevents you from continuing to use your post processing scripts, albeit with slight changes?

We are sorry for not disclosing this properly in the changelog. Let us know what do you think and whether this issue can be closed or not. Thanks is advance.

nmaggioni, 2024-03-04:
@SachCZ No problem, I was indeed already implementing your suggested solution. I just wanted to check if that was an intended behavior or a side effect of some other change! Thanks for clarifying.

j1m1sh, 2024-03-09:
I have a use case where I believe that this workaround does not work.

I use custom gcode to conditionally set the first level speed to 40 and then ramp up to 50, 60+
`{if layer_z <= first_layer_height}M220 S40
{elsif layer_z <= first_layer_height + layer_height * 1}M220 S50
...`

When printing multiple copies with "Complete individual objects" enabled, I would like the first layer of the subsequent objects to return to speed 40.

This is not custom gcode that is expected to run at just the start, but every time it returns to the first layer for each object.

j1m1sh, 2024-03-09:
Well, I just scrolled down and see a (new to me?) section call ed "Between objects G-code...".  It sounds like this and start is the place to put the first layer speed.

tg73, 2024-03-16:
I need to use custom gcode with gcode substitutions similar to how the XL homing tower works, but I need to use sequential printing and then apply the substitution at the first layer of each object. Now that the custom layer change gcode is not applied before layer zero, this is not possible. The workaround does not work, nor does "between object gcode", because the substitutions are processed per-layer-per-object. The substitution needs to happen at the start of each object, with visibility of the initial move to start the layer printing. Between objects gcode appears to be appended to the end of the previous gcode block for substitution purposes.

See https://www.printables.com/model/542670-homing-tower for this kind of approach - that is, custom gcode in tandem with gcode regex substitutions. In PS, load config from the gcode then look at the custom after layer change gcode, notably the `;;;` block. Then also look at the gcode subs, where:
```
^(;;;;BEFORE_LAYER_START\n(?:^;;;.*\n)+)((?:.*\n)*?)(G1 X.*\n)(G1 Z.*\n)
```
is replaced with
```
\4\2\1\4
```

In my use case, rather than waiting until layer_z is 5, the model is all in the first layer, has multiple objects printed in series to control the print order, and performs rehoming and other test work between objects.

@SachCZ please reopen this issue. For now I will have to revert to an older PS version.

SachCZ, 2024-03-28:
Hello @tg73, we will look into your use case. Thank you for your patience.

tg73, 2024-03-28:
Thanks @SachCZ . Here's an example 3mf as described above which relies on the combination of custom layer change gcode, sequential printing and gcode substitutions. This can only be sliced correctly in 2.7.1 or earlier.

[Issue12388_Example.zip](https://github.com/prusa3d/PrusaSlicer/files/14796465/Issue12388_Example.zip)

R2-G2, 2024-03-30:
I used the first layer for some extra fan settings. Now I have to move it to the second layer, which will work but it feels a bit strange ...

I had to put it there, because some variables are overwritten by the filament gcode (so the printer start gcode won't be enough).

```
;LAYER:[layer_num]
{
    if layer_num == 0 then
        if !ENABLE_CDS_FAN then
            "M107 P2\n";
        endif

        ENABLE_CHAMBER_FAN ? "M106 P1 S255" : "M107 P1";
    endif
}
{
    if print_settings_id =~ /^0\.20 Faster Temp Tower.*/ then
        local level = int(layer_num / FASTER_TEMP_TOWER_LEVELS_PER_LAYER);
        
        "M104 S" + FASTER_TEMP_TOWER_LEVELS[level] + " ; Faster Temp Tower @" + filament_type[0] + " on level " + level;
    endif
}
```

CapnBry, 2024-05-28:
I just updated from 2.7.0 to 2.7.4 and was confused that the Before/After Layer Change gcode wasn't running any more on the first layer.

There is a problem with the suggested replacement: that it runs before the filament gcode. My filament gcode includes `SET_GCODE_OFFSET Z=0.12` for ABS and `SET_GCODE_OFFSET Z=0.08` for PLA as the large bed temperature difference has slightly different Z offsets. If I move my hotend purge to the start gcode, it would purge before the Z-offset is set and scrape across the print bed.

I feel like this should be returned, and users who do not want the before/after gcode to run on layer 0 can just wrap their code in `{if layer_num != 0}`. There is no workaround for doing a hotend purge after the filament gcode now that the before/after gcode no longer fires on the first layer.

Is there somewhere else I can put my hotend purge that follows the filament gcode but comes before the first layer prints?

adrianhoehne, 2024-10-13:
For me, it was also a surprising behavior that the first layer change doesn't call "Before Layer Change" gcode. I would call a movement from Z0.0 to Z0.2(or whatever the first layer is) also a layer change. Therefore, I was expecting that the gcode would be executed before first layer starts.

In my case, I reduce the flow percentage in layer 0 and increase it in layer 1. So, I need now to copy that code across 2 sections, which can lead to confusion in later changes.

I would like to have at least an option that forces the "Before Layer Change" logic to be executed before the first layer also.

jwueller, 2025-06-29:
How about we just rename:

- "After layer change" to "Before layer"
- "Before layer change" to "After layer"

And then make it generate accordingly? That seems a lot more useful and solves the semantic issue at the same time. It would even be backwards compatible, because it's the same fields that are just being relabelled.

Currently, the only way to have consistent G-code per layer, for lack of an alternative, is to duplicate it into the "Start" and "End" G-code fields, which is quite error prone.

> [...] it is indeed **intended** as there is not really a layer change happening at the beginning of the print.

This doesn't seem quite accurate, since regular `LAYER_CHANGE` G-code is still generated at the start. Just not the user portion of it. Here is an excerpt from a file I just exported:

```gcode
;TYPE:Custom
PRINT_START ; ...
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;HEIGHT:0.2
G1 E-2 F1200
G1 Z.2 F15000
G1 X161.07 Y159.342 F30000
G1 Z.2 F15000
G1 E2 F900
M204 S5000
;TYPE:Skirt/Brim
;WIDTH:0.4
G1 F7200
```

To be clear: My proposed sequence would look like this instead:

```gcode
;TYPE:Custom
PRINT_START ; ...
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M107
;LAYER_CHANGE
;HEIGHT:0.2
G1 E-2 F1200
G1 Z.2 F15000
G1 X161.07 Y159.342 F30000
G1 Z.2 F15000
G1 E2 F900
M204 S5000
;BEFORE_LAYER   ; previously AFTER_LAYER_CHANGE
;TYPE:Skirt/Brim
;WIDTH:0.4
G1 F7200
; ...
;AFTER_LAYER   ; previously BEFORE_LAYER_CHANGE
;LAYER_CHANGE
;Z:0.4
;HEIGHT:0.2
;BEFORE_LAYER   ; previously AFTER_LAYER_CHANGE
; ...
;AFTER_LAYER   ; previously BEFORE_LAYER_CHANGE
;TYPE:Custom
PRINT_END
```

Note that this would probably also interact much better with sequential printing and tool change G-code, since a typical user would probably not intuit where exactly those fall in relation to layer changes, rather than layers themselves. Technically, any Z-movement like a Z-hop could be considered a "layer change", which would not be very useful either.

closed 2026-09-29 by prusaslicer-bot[bot]