prusa3d/PrusaSlicer #12389: No layers were detected.

JoJunYeong, 2024-03-04:
### Description of the bug

I'm trying to slicing using cli of prusa
as you can see, the file that I uploaded can't slicing at all
how can I solve it?
it driven me crazy recent few days 

[root@ip-172-31-29-225 ~]# /home/webapp/PrusaSlicer.AppImage --repair --ensure-on-bed -g "/home/webapp/stl_files/1.stl"  --load  "/home/webapp/slicer_profiles/detail100.ini" -o "/home/webapp/gcode_files/a.gcode" --loglevel=5 | tee "/home/webapp/log/slicer__log_tmp.log"
[2024-03-04 00:48:38.750778] [0x00007f97a0c0ec00] [debug]   TriangleMesh::repair() started
[2024-03-04 00:48:38.915524] [0x00007f97a0c0ec00] [debug]   TriangleMesh::repair() finished
[2024-03-04 00:48:39.071757] [0x00007f97a0c0ec00] [info]    Starting the slicing process. Resident memory: 36MB; Shared memory: 22MB; Private memory: 14MB; Peak memory usage: 70MB
10 => Processing triangulated mesh
[2024-03-04 00:48:39.072046] [0x00007f97a0c0ec00] [info]    Slicing volumes... Resident memory: 36MB; Shared memory: 22MB; Private memory: 14MB; Peak memory usage: 70MB
[2024-03-04 00:48:39.075371] [0x00007f97a0c0ec00] [debug]   slice_mesh to polygons
[2024-03-04 00:48:39.346375] [0x00007f97a0c0ec00] [debug]   Slicing volumes - removing top empty layers
[2024-03-04 00:48:39.348926] [0x00007f97a0c0ec00] [debug]   Slicing volumes - make_slices in parallel - begin
[2024-03-04 00:48:39.348957] [0x00007f97a0c0ec00] [debug]   Slicing volumes - make_slices in parallel - end
No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.

### Project file & How to reproduce

[0229_1.zip](https://github.com/prusa3d/PrusaSlicer/files/14475393/0229_1.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

amaon linux

### Printer model

mk3s+

rtyr, 2024-03-04:
Could you also please share the config you are using?

JoJunYeong, 2024-03-04:
@rtyr I just upload file on article!

JoJunYeong, 2024-03-05:
> Could you also please share the config you are using?

@rtyr I just upload file on article!

rtyr, 2024-03-05:
I was able to reproduce it (2.7.2 win64).

JoJunYeong, 2024-03-11:
@rtyr
how is it going?
is there any progress?

closed 2026-09-29 by prusaslicer-bot[bot]