prusa3d/PrusaSlicer #12390: 2.7.2 crashes when generating gcode

Anathema-Device, 2024-03-04:
### Description of the bug

After updating to 2.7.2 slicing crashes when generating gcode.  To reproduce the problem just load the provided project and slice.

### Project file & How to reproduce

[SD_card_box_table -Body-(Project).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14476225/SD_card_box_table.-Body-.Project.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

windows 11 23H2 

### Printer model

Prusa XL5T

SachCZ, 2024-03-04:
Hello @Anathema-Device, I have reproduced your issue in 2.7.2. It seems to be a problem somewhere in binary gcode generation. Currently, as a temporary workaround you may disable `Supports binary G-code` in `Printer Settings -> Firmware` and it seems to solve the issue.
Thank you for this bug report!

Anathema-Device, 2024-03-04:
OK, thanks, I will give that a whirl.  I was just about to give another example as it does not happen with all models.

ChiefPoints, 2024-03-10:
I'm seeing a similar issue with the MacOS version of PrusaSlicer 2.7.2, I've attached the project file that crashes. I'm not seeing this on other files I've attempted to slice.  Disabling "Supports binary gcode" makes no difference.

Since my problem is MacOS, instead of Windows, do you want me to open a separate bug report - or do the win and macos version share code that would cause this to appear on both operating systems? If so, let me know and I'll open a new bug report.

[PrusaSlicer_2.7.2Crashes.zip](https://github.com/prusa3d/PrusaSlicer/files/14551375/PrusaSlicer_2.7.2Crashes.zip)

SachCZ, 2024-03-13:
Hello @ChiefPoints, it seems to be a different issue. We will have to investigate it. Please be so kind and open a separate issue.

ChiefPoints, 2024-03-13:
Will do, heading out the door, but will try to get it done later today sometime.

closed 2026-09-29 by prusaslicer-bot[bot]