prusa3d/PrusaSlicer #12391: STEP File import error when contain revolve for spherical objects.

MartenKL, 2024-03-04:
### Description of the bug

When importing STEP-File into Prusa Slicer it missas part of a revolve that is a sphere. Happens in all models I have done where part of a circle meets rotation axis perpendicularly. To exclude export errors from SolidWorks I also imported  the STEP-file into 3ds Max 2024.

[image]
[image]
[image]
[PolishingBall.zip](https://github.com/prusa3d/PrusaSlicer/files/14480821/PolishingBall.zip)

### Project file & How to reproduce

1. File/Import/Import STL...
2. Select STEP-file with revolve

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11

### Printer model

Prusa i3 MK4

MartenKL, 2024-03-04:
If the end of the circle does not end perpendicular to axis this problem disappears. Like adding a small 45 degree (90 revolved) point or "bevel" the corner in the sketch between the axis line and the circle.

u89djt, 2024-03-04:
I'm having trouble with STEP files sometimes too (including yours), but I've just learned from trying importing your step file into Fusion 360 and re-exporting as STEP, that this seems to introduce changes to the data that allow the slicer to work with it.
Here's your PolishingBall step file imported to- then immediately exported from Fusion 360, and this imports to prusaslicer 2.7.2 successfully for me under windows 10, along with the resulting 3mf file:  
[2ndgenpolishing.zip](https://github.com/prusa3d/PrusaSlicer/files/14481441/2ndgenpolishing.zip)
[image]
Note that there are errors in the mesh that the slicer built that aren't visually apparent, and that slicing succeeds despite these.

closed 2026-09-29 by prusaslicer-bot[bot]