prusa3d/PrusaSlicer #12393: Error in the Open Source code for Prusa Slicer

RogerInHawaii, 2024-03-04:
### Description of the bug

I discovered an error in the Open Source code for the Prusa Slicer that causes errors with the Feedrate value and can also cause a related assert statement to fail.

I've been working on a project that modifies and adds new features to the Prusa Slicer. In particular it generates additional GCode to control the printer when the GCode file is used as input to the printer.

One of the features of the GCode protocols is that it is perfectly valid to include comments in the GCode file. You do that by preceding any comment with a semicolon, ';'. You can have a comment on a line by itself and you can have a comment on the same line as a GCode command by just adding a semicolon after the full GCode command and then adding the comment. You might, for example, do something like:

G0 E-0.400000; RCG Filament Retract

In that case it's a filament retraction command that I have added to the Prusa Slicer as part of the new functionality I'm adding to the Prusa Slicer. I add comments like this to help identify the new GCode in the resultant GCode file in order to assist me in debugging any potential problems during development. The "RCG" is my initials and the "Filament Retract" just helps me, or anyone else assisting in the development, to easily understand what the GCode does.

The particular problem, with that sample:

is that Prusa Slicer does some checks on the created GCode file and incorrectly thinks that "Filament" is a Feedrate parameter (it only check the 'F', not the whole word, and shouldn't even by checking text in the comment!). It subsequently divides the Feedrate variable value by 60. But worse, since it can encounter similar comments on subsequent GCode commands it KEEPS DIVIDING BY 60 over and over again, until the Feedrate variable value hits zero and THAT causes an assert failure.

I've attached a pdf document that fully explains the error in the Prusa Slicer source code. 

### Project file & How to reproduce

[Error in the Open Source code for Prusa Slicer.pdf](https://github.com/prusa3d/PrusaSlicer/files/14487066/Error.in.the.Open.Source.code.for.Prusa.Slicer.pdf)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1

### Operating system

Windows 11

### Printer model

irrelevant

SachCZ, 2024-03-05:
Hello @RogerInHawaii, we appreciate your effort. To me it seems to me, that you blame ` 
CoolingBuffer::parse_layer_gcode` for invalidly parsing gcode comments. Yet `parse_layer_gcode` never claimed to take valid gcode as input. It is an internal function of PrusaSlic3r, that depends on some specific input. Did you manage to reproduce the error in some of the precompiled slic3r versions? If so, be so kind and provide a 3mf, where we can reproduce it. If the invalid parsing is caused by your changes in the prusa slic3r code than it might be an issue in your understanding of the code base.

Side note: We at PrusaSlic3r also find the comments useful, that is why you can turn on `Verbose GCode` in `Print Settings -> Output options`. Make sure to have binary gcode (`Printer Settings -> General`) disabled if you use the `Verbose GCode` option.

SachCZ, 2024-03-20:
Closing, there does not seem to be an issue.

closed 2024-03-20 by SachCZ