prusa3d/PrusaSlicer #12398: Alternating seam position when using seam painting or other seam position options that are not random

mikecarper, 2024-03-05:
If I where to paint 2 lines for seam painting on a model I'd like to have the option where the seams are done in an alternating fashion. If what is getting printed is a Cube there are 4 corners A, B, C, D that could be used. With this option then Corners A & B would be used in an alternating seam.

```
 A *----* B
   |    |
 D *----* C
```

In this scenario, the process would start at corner A for the first layer, and then on the next layer it would switch to start at corner B, and so forth, alternating to help prevent any potential head crashes by distributing the starting points.

LuckyResistor, 2024-03-07:
If I get you right, would the final seam look like this?

```plain
Side View:

----*>---------
---------*>----
----*>---------
---------*>----
----*>---------
---------*>----
----*>---------
---------*>----

"----" = layer
"-*>-" = seam start point and print direction.
```

mikecarper, 2024-03-07:
That's one option for sure. On some models there are several spots where a seam can go that's well hidden. If there's a way to alternate between these edges so that seam starts are not stacked on top of each other that's what I'm hoping for. Similar to Random in the sense that the seams are not stacked, just better placed to make them not noticeable like Nearest (edge) does currently. Here's the cube idea showing the staggered seams on 2 edges

```
\*>-------------/
\-------------*>/
\*>-------------/
\-------------*>/
\*>-------------/
\-------------*>/
\*>-------------/
\-------------*>/
```

closed 2026-09-29 by prusaslicer-bot[bot]