prusa3d/PrusaSlicer #12401: no option for change installation directory (if PS already installed)

KubaMiszcz, 2024-03-05:
### Description of the bug

there is no such option, it remembers its current location somehow (in registry?) and always install it in it, also dont ask if folder is empty, just deletes all in it and install new version

### Project file & How to reproduce

install PS, and install/update it again

please just add checbox for custom installation like in every other installes

[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

win10

### Printer model

mk3s+

SachCZ, 2024-03-05:
Hello @KubaMiszcz, it indeed remembers where it is installed and does not ask again, **that is intentional**. The deleting is bit more nuanced than you state. It should delete the files only if there are no any extra user files. For example if you store your 3mfs in the slic3r installation directory it should never delete them. Let us know if you for some reason think it is not the case.

Also could you please elaborate why does it bother you, is there some use case where you want to choose the installation directory even after you have prusa slic3r already installed.

KubaMiszcz, 2024-03-21:
someday i point installation to dirA, now i have PS in dirA/PS, i dont know how i've changed it
in dirA i have many files like hints, presets etc, aaaaaand last time after update all is gone and i see PS file inside, lucky have cloud backup, few weeks ago when i try to update... same story, so it deletes files with no warning - it is unacceptable

your explanation is ok but where I can change it? registry?

and last reason: all installers i see have such option

SachCZ, 2024-03-28:
Hello @KubaMiszcz, the latest installer should never delete your files. We will double check that.

closed 2024-04-15 by SachCZ