prusa3d/PrusaSlicer #12403: Automatic color change not working

3designsprints, 2024-03-05:
### Description of the bug

I have 2 Anycubic Kobra 2 Pro and had an order for dual color keychains and after slicing i used the automatic color change (first time i tried it). In the slicer i could see the setting and in which layer the filament change would occur. I stayed near by, waiting to be notified or for the print to pause but nothing happened. By the time i realized that the print is past the color change layer it was too late and now i have 2 prints with lots of single color keychains i cant use... Why did this happen? Doesn't this feature work with the Kobra 2 Pro?

### Project file & How to reproduce

[color change gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14496500/color.change.gcode.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 11 version 23H2

### Printer model

Anycubic Kobra 2 Pro

rtyr, 2024-03-06:
PrusaSlicer only tells the printer to perform the color change, which is then usually handled completely by the firmware. M600 command is used by default, but you can specify any g-code you want in the printer settings.

I can see your printer is Klipper based, so you will likely have to set "M600" macro within your klipper cfg.

[image]

closed 2024-03-06 by rtyr

3designsprints, 2024-03-06:
Ok Thank you very much for your help!!

robandroll, 2025-03-18:
M600 works as expected with no need to press pause on Kobra 2 PRO (currently using firmware 3.1.4, but has worked for a number of firmware versions prior). Do NOT use prusa's auto-proposed color changes (if they exist). I'm unsure if this would be different on Kobra 2 non-Pro.

Color change G-code:
_M600
G1 E0.4 F1500 ; prime after color change_

[image]

Remove any Prusa auto-proposed color changes (I have it add them and then replace them for specific prints).

Right click on the layer where you want the color change to take place in Prusa and select the option "Add Color Change"

[image]

The printer will pause and move and hang indefinitely with nozzle and plate still hot. Here you can manually retract existing color, feed new color in, and then press the play button to resume the print. Immediately after resuming, it will extrude an ample amount of filament (to reduce, you can lower the 1500 in the g-code above), and then resume.