prusa3d/PrusaSlicer #12404: V2.7.2 For Mac OS Is Not Correct

aramperez, 2024-03-05:
### Description of the bug

I'm running v2.7.1 and I've downloaded, either via running PrusaSlicer or via a browser, V2.72 multiple times and copy it to my Applications folder (which notifies me that an existing copy, to which I select Replace).

When I run the "new" version, I get notified that a new version (2.7.2) is available.

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1+MacOS-x64

### Operating system

macOS 12.7.3

### Printer model

N/A

mix579, 2024-03-05:
Works fine for me. What does the top of the screen say, what version is it ? I suspect there's an old version sitting on your disk. Right click on the Prusaslicer icon in the dock and show on finder to find and delete it. you may have one version in the Prusa drivers folder inside the Applications folder and one in the Applications folder itself.

aramperez, 2024-03-05:
Thanks @mix579, I did have two versions. I don't know how because I always use the Installer and follow the instructions. Usually they are just to drag a folder to my Applications folder. IMHO, this is something that could be detected in the software.

closed 2024-03-05 by aramperez