prusa3d/PrusaSlicer #12407: Print Job Crashes Mk4

magoldsm, 2024-03-06:
### Description of the bug

I created a multi-color print for my Mk4 using 3 extruders (Multi-Material Single Extruder is checked).  The resulting gcode crashes my printer.  Screen is blank except for the time and the word Print in the upper left corner.  Knob has no effect.

I've attached the bgcode file and the 3mf file is the ZIP.

### Project file & How to reproduce

[Homebrew Mk4i Cover_0.4n_0.2mm_PLA,PLA,PLA_MK4IS_9h21m.zip](https://github.com/prusa3d/PrusaSlicer/files/14506731/Homebrew.Mk4i.Cover_0.4n_0.2mm_PLA.PLA.PLA_MK4IS_9h21m.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2+Win64

### Operating system

Windows 11

### Printer model

Mk4

SachCZ, 2024-03-06:
Hello @magoldsm, is this something that happens only with the latest Prusa Slic3r 2.7.2+Win64? I suspect you may be running into https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3525, which is a known firmware issue. Thanks for reporting it anyway! Please let me know whether I can close this issue in favor of the firmware issue.

magoldsm, 2024-03-06:
Reading that bug report, it appears to be exactly the same.

 

How do I roll back to working firmware / slicer?

SachCZ, 2024-03-06:
Slic3r is probably not the issue. So to rollback firmware, go to: https://help.prusa3d.com/downloads/mk4/firmware and download and flash version 5.0.1. In the meantime, feel free to upvote the mentioned firmware issue.