prusa3d/PrusaSlicer #12409: PrusaSlicer 2.7.2 release for macOS includes Prusa Drivers

nmattia, 2024-03-06:
### Description of the bug

The latest official release, https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2, contains "Original Prusa Drivers" instead of the usual ("PrusaSlicer" which can simply be copied to the Applications dir).

### Project file & How to reproduce

Download [2.7.2/MacOS-universal ](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.7.2/PrusaSlicer-2.7.2+MacOS-universal-202402291330.dmg) and open the archive. Unlike previous releases (like [rc2](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.7.2-rc2/PrusaSlicer-2.7.2-rc2+MacOS-universal-202402261536.dmg), 2.7.1, etc) it contains "Original Prusa Drivers" instead of just "PrusaSlicer".

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

MacOS-Universal

### Printer model

Prusa mk3s

nmattia, 2024-03-06:
[image]
Left: 2.7.2, right: 2.7.2-rc2

mix579, 2024-03-06:
All (or almost all) "final releases" have included the extra files in that folder, so it's not really a bug but a long-standing tradition 🙂

mix579, 2024-03-06:
Just copy the whole folder into the Applications folder (and make sure you delete the old Prusalicer.app file from Applications). It doesn't actually install any drivers, only two weblinks to sample models etc. Or open the Driver folder on the downloaded disk image and drag the Prusaslicer.app file into the Applications folder.

nmattia, 2024-03-06:
I had no idea you could open those files! But indeed, right click -> open and it opened a new Finder window. MacOS is still a mystery to me.

Thank you very much!

closed 2024-03-06 by nmattia