prusa3d/PrusaSlicer #12411: If fuzzy skin work like suport?

Junior109007, 2024-03-06:
**Describe the solution you'd like**
I think would be awesome if we could use Fuzzy skin like a Suport tool

**Describe how it would work**
As like suppot tool, we could paint some areas that we want to use fuzzy skin for a diferent type of surface, the possibilityes are endless, and if scalete with images for textures, would be like the next master top tool that only prusa has. 

**Describe alternatives you've considered**
Fuzzy skin if selectable area is already awesome.

**Additional context**
[image]
[image]

neophyl, 2024-03-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6988

Jan-Soustruznik, 2024-04-08:
Closed as duplicate https://github.com/prusa3d/PrusaSlicer/issues/6988

closed 2024-04-08 by Jan-Soustruznik