prusa3d/PrusaSlicer #12414: First layer infill speeds

slwise, 2024-03-07:
Setting the first layer speed slower helps with adhesion for the parameters but really slows print time doing the infill.

It would be great to be able to have one speed setting for the parameters and one for the infill. This feature is already available in Bambu and Orcaslicer. 

Here is a screenshot from Orcaslicer.
[image]

jdlawley, 2024-03-21:
Second

neophyl, 2024-03-21:
One thing you can do already is to set your first layer speed to a percentage.  That way it uses that % of the appropriate feature.  I use it all the time to get my first layer infill faster than the perimeters.

slwise, 2024-03-21:
Thanks. I'll give it a try.

TerPuc, 2025-03-06:
Added in PS2.9.1-alpha1. Closing.

closed 2025-03-06 by TerPuc