prusa3d/PrusaSlicer #12416: error rendering .step file

OtmarSchraml, 2024-03-07:
### Description of the bug

This error occurse befor the slicing process. The .step file was created from my own drawings by Fusion 360. Rendering the same model from .obj or .stl works fine.

### Project file & How to reproduce

Just import the .step file in any direction.
[stepError.zip](https://github.com/prusa3d/PrusaSlicer/files/14528775/stepError.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11

### Printer model

SOVOL SV03

u89djt, 2024-03-07:
The zip file you've provided doesn't contain a step file, just a 3mf file.
Looking closely at the model in the 3mf file, the surfaces become very gritty near the end of the piece with the big glitch. I've seen that happen sometimes. If you can provide the Fusion file, too, I'll have a look at it.
I'm just a user of the software, nothing to do with the repository.

OtmarSchraml, 2024-03-08:
Thanks for responding!

For your information,- Anycubic Slicer makes the same error as Prusa Slicer. Reimporting in FUSION 360 works fine.

[stepError.zip](https://github.com/prusa3d/PrusaSlicer/files/14533244/stepError.zip)

u89djt, 2024-03-08:
Right, when you import your STEP file into Fusion and then export /that/ as a new STEP file, it works find in the slicer. That says to me that the original construct in Fusion has numerically stiff surfaces in the original design that a mesh converter can have trouble with. Can you post the Fusion file itself - export as f3d - and I'll see if I can spot the problem?
In the image below, your original 3mf file is furthest from the camera, your original STEP file is in the middle, and re-export of the imported STEP file is closest to us with no glitches. That tells me that Fusion refactored the BREPs when it was importing the STEP because they were deemed likely to be troublesome. I think it will be obvious if we ask Fusion to convert your original design - not re-imported from the STEP file - into a mesh (tesselate under create in the mesh tab). 
[image]
 
(I'm not affilated with Prusa in any way)

OtmarSchraml, 2024-03-09:
[Downloads.zip](https://github.com/prusa3d/PrusaSlicer/files/14545716/Downloads.zip)

u89djt, 2024-03-09:
I don't think your choice of CAD design sequence caused it.
The only ways I've found to get a thread that can be exported to STEP then imported to the slicer successfully are
1) shorten the thread to just cover the region covered by the nut you're using (19mm works fine, for example), or
2) build the bolt out of shorter chunks:
Slice the long threaded bolt you made 12.5mm from the end before you chamfer it. That gives you a short threaded chunk with threads starting and ending at the same angle. Slice off the angled hex head (and delete the voids in its surface) and discard the long threaded region that the chunk was cut from. Create a rectangular pattern of the chunk you made along the path of the bolt, and then join those chunks together with the hex head. That looks like this: 
[rebuiltchunkedbolt.zip](https://github.com/prusa3d/PrusaSlicer/files/14546855/rebuiltchunkedbolt.zip)
(I haven't bothered to chamfer it.)
Speculation: long helices can cause numerical difficulties in STEP files and the mesh conversion process in prusaslicer if you don't break them up into shorter lengths.
Suggestion: make a lot of different threaded rods to find out more about the failure modes.
(not affiliated with Prusa)

closed 2026-09-29 by prusaslicer-bot[bot]