prusa3d/PrusaSlicer #12418: v2.7.2: supports being generated in mid-air

andyylin, 2024-03-08:
### Description of the bug

Whether using Snug or Organic supports, PrusaSlicer generates supports that are either disconnected from the layers below or generated in mid-air.

[image]
[image]

### Project file & How to reproduce

[1GF 縮小版-1L -3D 列印-2024.03.04  桶蓋一體列印+PD-203----提把.zip](https://github.com/prusa3d/PrusaSlicer/files/14534170/1GF.-1L.-3D.-2024.03.04.%2BPD-203----.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+win64

### Operating system

Windows 11

### Printer model

MK3S+ with MMU2S

morphias2004, 2024-05-10:
+1

I'm adding to this as I read through #7816 and #9601 and they appear to be different issues. As is #8609 .

I'm using PS 2.7.4 with the latest config updates.

When using Grid supports, I am experiencing instances of the supports and interface layers being generated in mid air.

Adjusting the pattern spacing, support extrusion width and XY separation have varying degrees of improvement, but typically make the issue worse or move the problem to a different part of the print.

PS Defaults:

[image]

And after some increasing the support pattern spacing from 2mm to 3mm, it's better:

[image]

Then increasing the support extrusion width from 0.36 to 0.38mm and reducing the XY separation from 80% to 60%:

[image]

[image]

morphias2004, 2026-09-18:
Over 2 years, no action and just gets closed. Is this fixed in PS3?