prusa3d/PrusaSlicer #12421: Fix by Windows repair algorithm causes sinking

FrHePo, 2024-03-08:
### Description of the bug

When using the "Fix by Windows repair algorithm" on a properly -on the build plate- placed object the object becomes a new alignment that is not congruent with the build plate anymore. Under the objects name the warning "sinking" is displayed.
The object must be "placed on face" again after using the "Fix by Windows repair algorithm".
I suspect that the special geometric form of my object is causing this misalignment after using the "Fix by Windows repair algorithm"

### Project file & How to reproduce

[Gehäuseteil rechts ASA.zip](https://github.com/prusa3d/PrusaSlicer/files/14538412/Gehauseteil.rechts.ASA.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+win64

### Operating system

Windows 10

### Printer model

Prusa MK3

u89djt, 2024-03-10:
Hi there, the base of the mesh in the 3mf file is not flat.
The slicer is aware of this, but has offered a face that is flat enough to use as the base.
You can just barely see hints of problems in the 3D editor view. It's more obvious if you run edge detection:
[image]
Importing the mesh into Fusion and converting to a faceted body then measuring some angles reveals lots of small problems.
[image]
Stretch that face and it's more obvious, and the repaired object has issues that behave similarly but slightly differently. The upper stretched surface is the original, and the lower is the repaired mesh.
[image]
It appears that the original object can be interpreted as flat enough to print on that face, and the slicer made a guess according to its criteria that predict something about successful slicing. The repair method tried to do something to make a better mesh in some sense, and in attempting to resolve the complex surface, returned an object with a differently shaped face. Some of those different tiny spikes project infinitesimally but definitely through the base of the printing model, so it's reported as sinking.
Quickest solution: align again and see if you're happy with the slicing outcome.
Better solution: find a way of creating a mesh with a cleaner, flatter base
Speculation: if you moved the part between different software packages one or more times, that might have lost some accuracy. The base surface seems to have a lot of different opinions about where it should be.
(I am not affiliated with Prusa. I like it when other people make my mistakes for me when I'm not in a hurry.)

FrHePo, 2024-03-16:
Hi,
Thanks a lot for checking this issue! I checked my .amf-file and I was actually a bit astonished to find the mentioned inaccuracies. The past years using SolidWorks I converted my designs into the amf-format for the use in PrusaSlicer. Now knowing about the inaccuracies I changed to the stl-format which solved the above mentioned problem.

FrHePo, 2024-03-16:
Issue solved.

closed 2024-03-16 by FrHePo