prusa3d/PrusaSlicer #12423: Slicer closing when slicing.

ke19lly, 2024-03-08:
### Description of the bug

When attempting to slice a threaded container I found on thingiverse prusaslicer closes. Ive tried various fixes found on google like running as admin, running windows 8 compatibility mode, trying both IS and non IS profiles for my MK4. Nothing is working.

### Project file & How to reproduce

[Closing.zip](https://github.com/prusa3d/PrusaSlicer/files/14543820/Closing.zip)
Just slice it and it will close the slicer.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.2+win64-202402291307

### Operating system

Windows 11 (build 22631), 64-bit edition

### Printer model

Prusa MK4

buzzhuzz, 2024-03-09:
Tried it out of curiosity on Linux version PrusaSlicer-2.7.2+linux-x64-GTK3-202402291307 and got it sliced succesfully.

However, my dev build out of git 4f035bceb actually crashes with assertion failed

> prusa-slicer: /home/dbuzdyk/packages/3d/PrusaSlicer/src/libslic3r/PerimeterGenerator.cpp:623: Slic3r::ExtrusionEntityCollection Slic3r::traverse_extrusions(const PerimeterGenerator::Parameters&, const Polygons&, std::vector<PerimeterGeneratorArachneExtrusion>&): Assertion `std::prev(it)->polyline.last_point() == it->polyline.first_point()' failed.

I hooked up GDB and upon inspection of extrusion_loop I see loop is indeed non-coninouos (order of polylines and points retained):
```
extrusion_loop = {
    polyline[0] = {
        { -9142803, -18732127 },
        { -9212051, -18716346 },
    },
    polyline[1] = { 
        { -9212051, -18716346 },
        { -9142803, -18732127 },
    },
    polyline[2] = {
        { -9212051, -18716346 },
        { -9533132, -18589963 },
    },
    polyline[3] = {
        { -9533132, -18589963 },
        { -9854214, -18463580 },
        { -10342131, -18170441 },
        { -10283691, -18449071 },
        { -10288628, -19248673 },
        { -9626535, -18800334 },
        { -9419293, -18758340 },
    },
    polyline [4] = {
        { -9419293, -18758340 },
        { -9212051, -18716346 },
    },
}
```

However, we can get closed loop after switching directions for polylines 0 and 1

I'd like to play with this task, but currently I'm not sure how to approach it

ke19lly, 2024-03-11:
> PerimeterGenerator::Parameters&, const Polygons&, std::vector&): Assertion

So I actually narrowed the problem down to my lga 1700 chip. If I set my
processor affinity for prusa slicer to only processors 0-7 it no longer
crashes.

lukasmatena, 2024-03-11:
Closing, this is now superseded by #12439, which you also created.

The assert mentioned by @buzzhuzz is very likely unrelated. We will investigate what causes it, though.

closed 2024-03-11 by lukasmatena