prusa3d/PrusaSlicer #12427: Unwanted move (wipe?) before retraction in external perimeter

seancmonahan, 2024-03-09:
### Description of the bug

PrusaSlicer emits a move (G1 with no extrusion) before the retraction at the end of external perimeters.

[image]

Expected would be no move before the retraction:

[image]

### Project file & How to reproduce

[PrusaSlicer - external perimeter unwanted wipes.zip](https://github.com/prusa3d/PrusaSlicer/files/14546750/PrusaSlicer.-.external.perimeter.unwanted.wipes.zip)

1. Disable `Wipe while retracting`
2. View G-code at the end of external perimeters.

The only non-default settings are disabling `Wipe while retracting`, and setting `Minimum travel after retraction` to 0 mm for clarity in the G-code visualization. Otherwise all default print, filament, and printer profiles for Ender-3 V2.

This occurs with both regular (`G1 E-`) and firmware (`G10`) retractions.

While reviewing the report before posting, I noticed that enabling `Verbose G-code` adds the comment `; move inwards before travel`. Is this a configurable setting? Would it be possible to disable it, or at least reorder it so the retraction happens before the inward wiping motion?

For now, I'm just adding a regex to my Output options to comment out moves immediately before retractions.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+linux-x64-GTK3

### Operating system

AppImage x64-GTK3 on Linux 6.7.4 #1 SMP PREEMPT_DYNAMIC Sat Feb 10 09:41:20 UTC 2024 x86_64 GNU/Linux

### Printer model

Ender 3 V2 - Klipper, but happens with default profiles, too

seancmonahan, 2024-07-22:
For now, I added the following G-code substitution regular expression in Print Settings -> Output options
Find:
`^(G1(?: [XYZ][0-9.]+)*( F[0-9.]+)?(?: ?;.*)?$)\n(G1 E-|G10)`
Replace with:
`;\1 ; unwanted wipe\nG1\2 ; extracted speed from elided wipe\n\3`
Description:
`remove unwanted wipes before retraction (see https://github.com/prusa3d/PrusaSlicer/issues/12427)`

staal54a, 2024-07-25:
+1 also seeing this issue

closed 2026-09-29 by prusaslicer-bot[bot]