prusa3d/PrusaSlicer #12428: Ramping Z Lift causes absolute E to reset to 0 upon activation

FasterDBConsulting, 2024-03-09:
### Description of the bug

When the Ramping Z Lift feature is activated in combination with a printer using Absolute E values, the E value is reset upon reaching the first layer where z lifts are allowed. This causes a print failure for Klipper which uses absolute E.

### Project file & How to reproduce

[PS Ramping Lift Bug Report.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14547109/PS.Ramping.Lift.Bug.Report.3mf.zip)

[image]
[image]

1. Uncheck Use Relative E distances under Printer Settings > General
2. Remove any G92 E0 from custom gcode sections
3. Under under Printer Settings > Extruder 1: Check Use Ramping Lift. Set Max Ramping Lift to 0.2, Only Lift Above Z 0.2, and Retraction Length to 0.4
4. Slice and inside the gcode, find the first layer change that is >= Z0.2.
5. Note that in gcode after the layer change, the E values have begun counting from 0 again.

The result of this reset during printing, e.g. in Klipper, is an error similar to `Extrude only move too long (-525.213mm vs 50.000mm)`

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+win64

### Operating system

Windows 10 x64

### Printer model

voron v2.4

FasterDBConsulting, 2024-03-09:
Closing because my provided example doesn't demonstrate the issue correctly and I'm still working out the what caused G92 E0 to not be sent during layer change.

Jan-Soustruznik, 2024-03-25:
Hello, @FasterDBConsulting , thank you for your report. Problem is now fixed in latest alpha release of 2.7.3, link to download: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1 

Closed as solved.

closed 2024-03-25 by Jan-Soustruznik