prusa3d/PrusaSlicer #12430: Constrain cut to X, Y or Z (for precise plane cuts)

Stooovie, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
The Cut feature lacks an obvious way to constrain the cutting plane, making precise cuts impossible.

**Describe the solution you'd like**
Almost all graphical applications use Shift to constrain drawing lines. Since Shift is already used to draw the cut, a different key would be used for constraining. Or, snapping could be enabled by default as most people probably use straight X, Y or Z-plane cuts  anyway.

neophyl, 2024-03-09:
The angle of the cut plane is (or rather I should say CAN) be constrained already.  The direction arrows in combination with the compass can be used to snap it to precise angles.  Its the exact same tool process used to rotate things.

[image]
For example after grabbing the handle if you position the mouse over the inner marks of the compass you can snap to 45 degree increments, over the outside you get 5 degree steps and moving the mouse outside the compass allows free rotation.

Stooovie, 2024-03-09:
It's weird, I was not getting the cut after activating the tool unless I drew the cut manually with Shift...

EDIT: it's a semi-user error, coming from Blender I'm used to zooming in pretty close, and that made the default cut and the gizmos not visible in the viewport.

Look: 
[image]

This is immediatelly after activating the Cut tool. I'm lead by the UI to start manually adding the cut, as the default one is not visible. I'm definitely in the minority here but the tool should be always apparent.

neophyl, 2024-03-09:
Yeah the zoomed in losing the widgets is a known issue.  Theres this one https://github.com/prusa3d/PrusaSlicer/issues/11344
I think its also a regression too as I also found this closed one https://github.com/prusa3d/PrusaSlicer/issues/2028 reporting the same thing thats marked as completed.

closed 2026-09-29 by prusaslicer-bot[bot]