prusa3d/PrusaSlicer #12431: paint on support does not work with negative volume

robomaniac, 2024-03-09:
### Description of the bug

I notice that paint-on-support is affected by negative volume. You can not add paint-on-support when you added a negative volume.

I added object, right click and add negative volume: cylinder
[image]

I paint-on-support but you can see once slice, the negative volume does not work
[image]

I then decide to check with support everywhere and you see it works with negative support
[image]
[image]

I redid this entire step-by-step in a new window to double check. My part is much smaller and same behaviour. I scaled this up real big to make sure everything is clear.

I thought this was similar to issue #11804 but no

### Project file & How to reproduce

[paint-on-support_negative_volume.zip](https://github.com/prusa3d/PrusaSlicer/files/14547507/paint-on-support_negative_volume.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

windows 10

### Printer model

Prusa MK3.5

neophyl, 2024-03-09:
When you use a paint on method you are defining surfaces 'on the model geometry'  that need support.  When you used the negative modifier to stop the slicer adding any extrusion into that area those 'faces' on the model don't really exist, so naturally you can't paint them to define them as needing support.  The negative modifier is making no changes to the underlying model geometry, its a modifier to the slice produced.

When you use support everywhere the software decides based on the angles of surfaces if it needs support (based on your support settings).  So it can programmatically add support to the areas effected by the modifier. 

Its a limitation of modifiers and painting when used in conjunction.

robomaniac, 2024-03-09:
Thanks for the explanation, can this be a features request then?Thank you,Jérôme

RanTheLab, 2025-06-18:
What I have done is use supports 'everywhere', then paint on support **blockers** where I did not want supports on any other parts of the model.  This results in supports where the model had negative volume removed, and no supports where the blockers were.

closed 2026-09-29 by prusaslicer-bot[bot]