prusa3d/PrusaSlicer #12432: Avoid Crossing Perimeters not working

Nicknomadic, 2024-03-09:
### Description of the bug

Avoid Crossing perimeter function does not work in 2.7.2. Works perfectly fine in 2.7.0. did not test any other version.

### Project file & How to reproduce

[Right Ride Armor.zip](https://github.com/prusa3d/PrusaSlicer/files/14547567/Right.Ride.Armor.zip)
1. click on Print settings
2. go to Layers and perimeters
3. check on Avoid Crossing perimeter
4. go back to Plater 
5. click Travel in the legend
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+win64

### Operating system

Windows 10 Pro 22H2

### Printer model

Original Prusa i3 MK3S+

dwbmb, 2024-03-09:
I just came here to post the same issue.

### Description
For me, avoid crossing perimetres increases fast travel lenght by 50% somehow while still crossing perimetres sort of the same. Interestingly the issue disappeared after changing perimetre generator from Arachne to Classic. See pictures:

Default settings, Arachne, Avoid crossing periemtres turned off:
[image]

Turned on Avoid crossing periemtres, travel lenght set to 1000% (for sure), still Arachne:
[image]

Turned on Avoid crossing periemtres, travel lenght set to 1000% (for sure), periemtres generator switched to Classic:
[image]

### Files included:
3mf project: [ikea_samla_clasp_for22L.zip](https://github.com/prusa3d/PrusaSlicer/files/14547791/ikea_samla_clasp_for22L.zip)

### Version of PrusaSlicer
Version 2.7.2+win64

### Operating system
Windows 11 Pro 23H2

### Printer model
Prusa Mini+

Prizmagnetic, 2024-03-12:
I'm having a similar issue. Avoid crossing seems to be working except for right after the layer change. These crossing travel moves are the first move in each layer. Pic shows Arachne generation and classic just rearranges the positions. 

Version 2.7.2
Windows 11
Original Prusa i3 MK3S+

[image]

PabloAVazquez, 2024-03-12:
I'm having the same situation, avoid crossing perimeters, nearest seam, 600mm maximum detour, 2.7.1+ produces the following:
[image]

and using exactly the same profiles, 2.7.2+ produces the following:
[image]

tried switching to classic perimeter as dwbmb mentions but didn't fix it entirely
[image]

thanks!!!

lukasmatena, 2024-03-25:
Should be fixed in [2.7.3-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-beta1). Closing.

closed 2024-03-25 by lukasmatena

Trydnt, 2024-07-20:
@lukasmatena

This issue still persists in 2.8.0.
It isn't working because it doesn't consider all perimeters, only the external ones. This means it often makes travel moves within the perimeters creating stringing there which ruins the walls integrity and/or appearance. I've seen like 10+ tickets for this same issue using different terms but it's all the same issue. Please make it avoid crossing perimeters (PLURAL) instead of avoid crossing external perimeter.

Perhaps it would be better to have a minimum distance to maintain from perimeter during travel. (it can ignore this for the beginning and end of the travel move if it's beginning or ending that close to a perimeter.)

To make matters worse this is having really bad impact on travel optimization too. I have a design with several holes in it in a line. When moving past them it goes around each individually creating a drastically longer pathway. (Imagine the letter B. The I is the direct path. D would be the optimal path. But it's making a B shape for every single hole. Essentially it's always trying to stay on it's original path and deviates to avoid but there's no intelligence to the detour.

dwbmb, 2024-07-22:
I agree it is not that mess as it used to be but does not work entirely to date. Please reconsider this output @lukasmatena.

I am having these unnecessary crossings even with Avid crossing perimeters set to 10000%

[image]

Orafilynie, 2024-09-29:
I can confirm that on the 2.8.1 release version, the option is still broken, i have tested to downgrade to the 2.7.0 and it works perflectly.

I tested with the exact same parameters and stl files.

But i saw that the avoid crossing breaks when i set the seam option to aligned, it works when the seam option is set on nearest.

3dpolyex, 2024-10-24:
Avoid Crossing Perimeters not working (PrusaSlicer 2.8.1).

Unfortunately, this issue was marked as closed even when the problem persisted (why?). Is it possible to reopen it and try to find a solution?

I have contacted Prusa, and their only solution was to downgrade to 2.7.0 to get the job done. I hope the developers fix the issue.

According to its history on many different websites and forums, this issue has been around for a long time with no action taken.

**It should be considered serious because it affects the final results of the 3D-printed parts and the quality that Prusa guaranty to the customers.**

Gilles03300, 2024-11-26:
Actually the option to avoid crossing perimeters does not work correctly on 2.8.1
Do you know how you can install several versions of Prusaslicer on Windows? Because I would like to use several versions at the same time to test.
Thank you in advance for your help

JustJuan2024, 2024-11-26:
> Actually the option to avoid crossing perimeters does not work correctly on 2.8.1 Do you know how you can install several versions of Prusaslicer on Windows? Because I would like to use several versions at the same time to test. Thank you in advance for your help

Good morning, Gilles03300. I have been working with Version 2.8.0, which performs the job correctly. Most of us (users) hope the developers see this issue and try fixing it for the next release (2.8.2 ...?). Regarding your question, I don't know if it is possible or convenient to do so; I suggest working with the later version possible to have access to the new options. Consider that each new release comes with improvements to the software in general and slicing capabilities.

Orafilynie, 2024-11-27:
> Actually the option to avoid crossing perimeters does not work correctly on 2.8.1 Do you know how you can install several versions of Prusaslicer on Windows? Because I would like to use several versions at the same time to test. Thank you in advance for your help

Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1), as linked in the issue #13445 :)

Gilles03300, 2024-11-29:
je vais tester, merci @Orafilynie

Oxmaster, 2025-01-02:
This still seems like an issue, even with 2.9.0 
[image]