prusa3d/PrusaSlicer #12433: Organic supports breaking due to anomalous single wall layers generated at branch forks

loudaslife, 2024-03-09:
### Description of the bug

I've been trying to print a large animal figure in PVB using organic supports. On my most recent attempt, I had a print fail due to a snapped support branch, and I noticed something odd at the particular layer (Z: 20.4mm) where the break happened

[image]
[image]

It's a bit hard to see in the 3D view, so here are the layers immediately around the area of the break:

[image]

As you can see, the areas above and below the fork have double walls, as expected with the default organic support settings. But the last layer before the fork only generates a single wall, which almost entirely defeats the purpose of having double walls in the first place. The expected behavior would be to generate double walls for every layer that exceeds the "Branch Diameter with double walls" setting.

I looked around and found at least 7 areas in the support structure for this print with this defect:

Front support at Z: 65.3
Front-middle support at Z: 21.6
Middle-rear support at Z: 7.2
Middle-rear support at Z: 21.5
Rear support at Z: 10.2
Rear support at Z: 20.4 (this is the one that actually failed during the print and is shown in the above images)
Rear support at Z: 22.2

All of them occur in the same way, with only a single wall generated at the last layer before the fork splits.

### Project file & How to reproduce

Here's the project file where the defects were observed and which generated the failed print:
[v2 capybara 125mm mk4.zip](https://github.com/prusa3d/PrusaSlicer/files/14547797/v2.capybara.125mm.mk4.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Windows 11

### Printer model

Original Prusa MK4

loudaslife, 2024-03-09:
Additional testing I've done:
Replicated the issue on versions 2.7.1, 2.7.0, and 2.6.1.
Replicated the issue at the default support material extrusion width value of 0.35 mm.
Replicated the issue at 0.2mm layer height.
Replicated the issue with more conservative values for "Maximum Branch angle" and "Preferred Branch angle"

I've noticed a few instances where the defect occurs at the second-to-last layer before the fork, or at the first layer _after_ the fork. So the exact vertical position can vary slightly.

loudaslife, 2024-03-14:
I've fiddled around some more, and at this point I can't come up with a configuration that *doesn't* replicate the issue. I installed PrusaSlicer on a different platform (Linux), imported a profile for a completely different printer, loaded a simple test model, enabled organic supports (without touching any other settings), and sliced it. Sure enough, I immediately found the defect again.

[image]

Project file: [support test.zip](https://github.com/prusa3d/PrusaSlicer/files/14606859/support.test.zip)

I think this issue is probably quite widespread, people just don't notice it because it's difficult to spot in the gcode preview and because it doesn't cause failed prints if your material has strong layer adhesion. Even when it does cause failed prints, you have to look somewhat closely to figure out the cause, so most people would attribute it to bad luck or think organic supports are intrinsically weak.

u89djt, 2024-03-14:
I'm seeing the same in everything I tried just now. Looking higher up the supports, we also see brief moments of double walls during generally single-walled regions as they go through a fork. It perhaps suggests that estimation of the section size for a test against "Branch Diameter for double walls" is glitching as the pair of notional circles gradually merges or splits when the circles are approximated with extrusion paths with extra little ticks. Unsurprisingly difficult.

rob-miller, 2024-04-26:
I can't say if this is the reason my organic supports have broken, but printing 0.25 nozzle / 0.05 layer height on XL I have yet to be able to complete a print and often because the organic support breaks.  

While the specific issue here seems like a bug, in addressing it I would like to see a way to make organic supports stronger.  Perhaps some infil, or perhaps more perimeter layers.

closed 2026-09-29 by prusaslicer-bot[bot]