prusa3d/PrusaSlicer #12438: Z-Offset

Aaron1073, 2024-03-10:
**Is your feature request related to a problem? Please describe.**
I use all kinds of filament not just Prusament and I find the flow for all them are different as well as the squish changes with them, making it almost imposable to remove the print from the bad.

**Describe the solution you'd like**
I would like to see a z-offset per filament

**Describe how it would work**
just like adding a new custom filament just adding a z-offset field.

**Describe alternatives you've considered**
anything

sarusani, 2024-03-10:
That won't solve the problem. If it's pushing out too much filament on the first layer, it will also do it on the other layers. You need to adjust the extrusion multiplier for your filament.

rtyr, 2024-03-11:
Also a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2744.

closed 2024-03-11 by rtyr