prusa3d/PrusaSlicer #12439: Implement A Function for Compatibility with LGA 1700 Processors

ke19lly, 2024-03-10:
**Is your feature request related to a problem? Please describe.**
When slicing large/detailed models the slicer crashes and will sometimes hard crash my PC.

**Describe the solution you'd like**
I found that setting processor affinity and only allowing the slicer to use cores 0-7 fixes this. But that method has to be done every time I open prusa slicer.

**Describe how it would work**
If possible have a setting with a checkbox for a sort of compatibility mode would be amazing. 

**Additional context**
My PC has a 14900k running at intel xmp clocks with 64gb of DDR5 6000 M/T ram.

Thank you for any time and consideration on this matter. Im loving Prusaslicer but this can be a major headache and is the one and only issue I have with the software.

u89djt, 2024-03-10:
You can create a shortcut to run a program with predefined core affinity: https://www.windowsdigitals.com/how-to-set-process-cpu-affinity-priority-permanently-in-windows-10/
(I'm not affiliated with Prusa.)

ke19lly, 2024-03-10:
Thank you I really appreciate the help I'll give that a try!

lukasmatena, 2024-03-11:
@ke19lly This sounds like another incarnation of https://github.com/prusa3d/PrusaSlicer/issues/11845. If you agree, could you post your solution into that thread and close this one, so we have all the information in one place? Thanks.

ke19lly, 2024-03-11:
Done.

closed 2024-03-11 by lukasmatena