prusa3d/PrusaSlicer #12440: Different infill pattern

AndylgTom, 2024-03-11:
### Description of the bug

How can I fix it so that the 100% fill in this example prints the same as in version 2.5? I don't like the way it is done now in version 2.7. Is it some new specificity of the new version or a bug?

### Project file & How to reproduce

Version 2.5
[image]

Version 2.7.2
[image]

File project.
[11.zip](https://github.com/prusa3d/PrusaSlicer/files/14558867/11.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.7.2

### Operating system

Win 10

### Printer model

never mind

neophyl, 2024-03-11:
Yes it is intentional and no , at the moment there is no way to stop it doing so.

Existing issues 
https://github.com/prusa3d/PrusaSlicer/issues/10573
https://github.com/prusa3d/PrusaSlicer/issues/10677  
https://github.com/prusa3d/PrusaSlicer/issues/11202
Probably others too.

AndylgTom, 2024-03-13:
I hope this will be fixed because the strength of the part is greatly reduced.

closed 2026-09-29 by prusaslicer-bot[bot]