prusa3d/PrusaSlicer #12442: Bad geometry after cut operation on simple model

eparadis, 2024-03-11:
### Description of the bug

I loaded a 3MF file produced by OpenSCAD and did a cut operation in PrusaSlicer. One of the resulting two halves has bad geometry that PrusaSlicer detects and will then not slice correctly.
﻿
[image]
[image]
[image]

### Project file & How to reproduce

1. Open the project file from before the cut operation
2. Select the object
3. Click the Cut tool icon
4. (it defaults to mid-model, which is the correct location)
5. Add connectors around the perimeter as shown
6. Select "flip upside down" for Object A
7. Click "Perform cut"
- Expected: the object is split in half with correct geometry and can be sliced and printed
- Actual: the object appears cut correctly in 3D Editor view, but the object list shows errors, and slicing view shows open edges.

Notes:
- even with no connectors, the model geometry problems will occur
- sometimes the cut operation results in 18 open edges and sometimes only 3. I don't know why.

before the cut operation:
[manifold_2x1_25hex - before cut operation.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14562275/manifold_2x1_25hex.-.before.cut.operation.3mf.zip)

after the cut operation:
[manifold_2x1_25hex - bad cut.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14562254/manifold_2x1_25hex.-.bad.cut.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+MacOS-arm64

### Operating system

MacOS Sonoma 14.3.1

### Printer model

Prusa MINI+

u89djt, 2024-03-11:
Windows user having a go: both files work without errors or missing solids for me in 2.7.2 under windows 10. I didn't have the mini profile as part of my standard settings so it was temporarily installed by the slicer for each of these files. Here's a project file of the uncut object imported into the slicer, then cut and connectors added, and the result saved. 
[u89djtcutofmanifold_2x1_25hex - before cut operation.zip](https://github.com/prusa3d/PrusaSlicer/files/14564169/u89djtcutofmanifold_2x1_25hex.-.before.cut.operation.zip)

eparadis, 2024-03-11:
I closed PrusaSlicer and opened the "before cut operation" version and repeated my steps because I wanted to capture a screenshot of the case when it only has 3 open edge errors. But... it magically fixed itself? Here is the project at that point:
[manifold_2x1_25hex - after cut operation - magically fixed.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14565084/manifold_2x1_25hex.-.after.cut.operation.-.magically.fixed.3mf.zip)

Hopefully someone can look at the various versions included and determine what sort of subtle situation is happening here.

closed 2026-09-29 by prusaslicer-bot[bot]