prusa3d/PrusaSlicer #12445: First layer change missing G92 E0

chdgriff, 2024-03-11:
### Description of the bug

I'm trying to print the Feet - DC version.stl from [printables link](https://www.printables.com/model/741704-dual-filter-solder-fume-extractor-hepa-carbon-acdc) but keep getting a "Extrude only move too long" thrown by Klipper (Mainsail).

I've figured out the issue to be a missing G92 E0 and possibly wipe on the layer change before layer 2.  This causes the subsequent extrusion command to throw the extrude too long error:

First Layer change (Before Layer 2):
[image]

Second Layer Change (Before Layer 3):
[image]
_Note the Wipe and G92 E0 missing from the first layer change in the first image._

So far, this is the only model I've had this issue with and the fact that it's only occurring on the first layer change seems to indicate that this is a bug.

### Project file & How to reproduce

[Feet - DC version.zip](https://github.com/prusa3d/PrusaSlicer/files/14563757/Feet.-.DC.version.zip)
Steps to Reproduce:

1. Slice File as is.
2. Use Gcode viewer to view the second layer and confirm no wipe or G92 E0 was generated at the start of the layer.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.2+win64-202402291307

### Operating system

Windows 11 (build 22631), 64-bit edition

### Printer model

Modified Ender 3 v2 running klipper

chdgriff, 2024-03-11:
I realized in manually inputting the missing `G92 E0` that it's also missing `G1 E.3 F1500` (configured retraction distance) as well

chdgriff, 2024-03-12:
If anyone has similar issues, I've made a quick and dirty python script: [fix_missing_gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14565619/fix_missing_gcode.zip) to programmatically add back in the missing extruder resets and deretractions for a given gcode file.

You made need to install tkinter if you opted out of it during your python installation.

rtyr, 2024-03-12:
This is the same issue as https://github.com/prusa3d/PrusaSlicer/issues/12428. I am closing this as a duplicate.

For now, you can either use relative E distances or disable ramping z-lift.

closed 2024-03-12 by rtyr