prusa3d/PrusaSlicer #12446: variable layer height stops working with custom printer

RS-RDS, 2024-03-11:
### Description of the bug

when i added a printer profile(given by the manufacturer of my printer) the variable layer height stops working.
and when i get the any prusa printer, the tool came back to use.

### Project file & How to reproduce

[variable layer height problem.zip](https://github.com/prusa3d/PrusaSlicer/files/14564783/variable.layer.height.problem.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2+ win64

### Operating system

win10

### Printer model

gtmax3d a2v2

mix579, 2024-03-11:
In Prusaslicer>Printer Settings>General, you have to enable variable layer height. Last option on that screen. You have to be in Expert mode to see it.

rtyr, 2024-03-12:
Answered.

closed 2024-03-12 by rtyr