prusa3d/PrusaSlicer #12447: External Perimeters First except on overhangs

Sable147, 2024-03-11:
**Is your feature request related to a problem? Please describe.**
The setting 'external_perimeters_first' is helpful for better dimensional accuracy, and for internal features not to shine through. However, it makes overhangs more difficult to print in my experience.

**Describe the solution you'd like**
Add a setting to allow external perimeters first on normal surfaces, but print internal ones first on overhangs.

**Describe alternatives you've considered**
The 'external_perimeters_first' setting could be changed by the user using height range modifyers, if that would be added (see https://github.com/prusa3d/PrusaSlicer/issues/3635), but that would still be a hassle.

CCS86, 2024-03-15:
I have also thought about this. The easiest implementation would probably be to default to ext. first, unless a layer has any region exceeding an overhang threshold, then switch to inner first. Trying to only switch the wall order locally around the overhang would lead to extra seams and would probably be really tough to implement.

timur-tabi, 2025-01-02:
IMHO, this is a really big deal that could greatly improve the default prints on every Prusa.  Printing external perimeters first provides a HUGE improvement in print quality, but God help you if you have an overhang.  Then you are screwed.

I have a print that looks gorgeous up until it gets to the top.  Even with organic supports, the overhang looks terrible.

I tried using modifiers to change the perimeter order for just a region of the print, but I just couldn't get it right.  It's only a few places that need the order changed.

timur-tabi, 2026-09-03:
This is still a useful feature to have.

coreyfro, 2026-09-18:
I support this feature request and vote to keep it open.

I print miniatures on FDM and have settings that rival SLA. This feature would sharpen features dramatically.

https://www.printables.com/model/1208569-prusaslicer-settings-for-miniatures-mk4-025mm-nozz