prusa3d/PrusaSlicer #12453: Right clicking color change does not show color picker but block entire slicer UI

makreljohnson, 2024-03-12:
### Description of the bug

I have a simple model witha color change half way up. The colors are blue and purple for the 2 colors. I wanted to change the purple to white to represent the finished product but when I right-click the colroed slider segment, the ui stops responding to interaction and Windows 11 emits the tone that a modal is open and I'm trying to access what's behind it. If I press escape, the ui begins responding again. The color picker modal is not visiable anywhere on the screen and does not appear to be hidden. by other windows. The behavior happesn in any window sonfig: Maximized snapped left or right or just in the middle.

This worked once to get the color set to purple (accidentally) but not since.

Prusa Slicer:  Version: 2.7.2+win64
Edition	Windows 11 Pro
Version	23H2
OS build	22631.3235
Experience	Windows Feature Experience Pack 1000.22687.1000.0

### Project file & How to reproduce

Slice model.
Hover over the colored slider at the second color so the "Edit current color - right click..." displays
Right click
For me, colopr picker does not appear.
I have uninstalled, deleted profile files and reinstalled. I suspect it's the project file...
[Token 1.zip](https://github.com/prusa3d/PrusaSlicer/files/14577666/Token.1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.7.2+win64

### Operating system

Windows 11 23H2 OS build	22631.3235

### Printer model

CR-10 (original)

neophyl, 2024-03-13:
Sounds like https://github.com/prusa3d/PrusaSlicer/issues/10419 is not a successful fix.

makreljohnson, 2024-03-13:
I just ditched the file and started over. The original must have gotten corrupted when I was messing with the color change. The second attempt has had no similar issues - just that the printer didn't hold the z height during filament change and ran into the lower layers. Oh well.

closed 2026-09-29 by prusaslicer-bot[bot]