prusa3d/PrusaSlicer #12455: build error: integer value is outside the valid range of values for this enumeration type [-Wenum-conste xpr-conversion]

nmattia, 2024-03-12:
### Description of the bug

I'm getting build errors:

```
PrusaSlicer/src/libslic3r/Config.hpp:342:35: error: integer value 4294967295 is outside the valid range of values [0, 15] for this enumeration type [-Wenum-constexpr-conversion]
```

This seems to be related to this template which uses the enum's underlying type's max value for "nil", which the compiler does not like:

```cpp
// For enums the nil is the max value of the underlying type.
template<class T>
struct NilValueTempl<T, std::enable_if_t<std::is_enum_v<T>, void>>
{
    using NilType = T;
    static constexpr auto value = static_cast<T>(std::numeric_limits<std::underlying_type_t<T>>::max());
};
```

I am building with this version of clang:

```
$ clang --version
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

The issue can be worked around by suppressing the warning:

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 3b37dbac6..594708cd4 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -244,6 +244,7 @@ if (NOT MSVC AND ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU" OR "${CM
AKE_CXX_COMP
         set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall" )
     endif ()
     set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-reorder" )
+    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-enum-constexpr-conversion" )
 
     # On GCC and Clang, no return from a non-void function is a warning only. Here, we 
make it an error.
     add_compile_options(-Werror=return-type)
```

See also the discussion on the clang tracker, which plans on making this a hard error: https://github.com/llvm/llvm-project/issues/59036

I'm happy to do some testing and submit a PR, but not sure what the best way to fix this is! Thoughts?

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Commit: https://github.com/prusa3d/PrusaSlicer/commit/cfbbd443b15a96c2e9bf9cd4670b4dbfe37ec04d

### Operating system

macOS 14.4

### Printer model

N/A

SachCZ, 2024-03-13:
Hi @nmattia I see the issue. We will try to fix it! Just suppressing the warning is not great. The issue lies in the fact that NilType is instantiated with `enum` an then not used anyway. We have a more systematic solution using SFINAE in mind. Thank you for pointing this out!

nmattia, 2024-03-13:
No worries! I have very little C++ experience but let me know if I can help!

nmattia, 2024-03-20:
Thank you @lukasmatena & @SachCZ !

SachCZ, 2024-03-28:
We had to revert the fix.

Salamandar, 2024-04-11:
> We had to revert the fix.

Hi, can you explain why you had to revert it ? Were there some unexpected side effects ?

SachCZ, 2024-04-11:
Hello @Salamandar, yes. I thought the repair commit should not change any runtime behavior, but for some currently unknown reason, it broke the "scale" default values of SLA print objectes in platter. There are multiple possible explanations. Most probably  the commit actually broke some runtime behavior without me understanding how. Other less probable possibility is that there is already some UB, but it is currently masked/mitigated somehow.

closed 2026-09-29 by prusaslicer-bot[bot]