prusa3d/PrusaSlicer #12462: Regression: "Arrange" function got worse

RichiH, 2024-03-14:
### Description of the bug

While Arrange was never perfect, it used to be better at autoplacing than the current version. See the automatic arrangement:
[image]

versus manually arranging:
[image]

or even:
[image]

While I understand that binpacking is an NP-hard problem, there's a trivial approach if the objects are all the same: Try quadratic arrangement as per photo or ideally offset as per second screenshot. The same could be tried for other similarly sized objects, first.

### Project file & How to reproduce

Import a bunch of STLs and click arrange

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.2+linux-x64-GTK3-202402291307.AppImage

### Operating system

Debian unstable

### Printer model

Prusa XL

RichiH, 2024-03-14:
CC @pstrobl96 because you asked for this

pstrobl96, 2024-03-14:
I have tried multiple older versions of PrusaSlicer. Obviously algorithm is non-deterministic and it's struggling when multiple smaller prints are on the bed. 

[image]
[image]
[image]
[image]
[image]
[image]

RichiH, 2024-03-15:
As an update, the "I didn't have this problem before" stuck in my head... After more debugging, it does not seem to happen with the Mk4 or Mini, only the XL. So maybe the total size is part of the issue?

u89djt, 2024-03-20:
What happens if you place a single instance on the bed, then right-click and select "fill bed with instances"?

u89djt, 2024-03-22:
I guess with the popularity of maker coins, it might be worth hard-coding the coordinates of good packings available for circles in a square for the XL plate.

This site provides coordinates for packing up to 5000 circles into a square in a handy format "All coordinates of all packings":
http://hydra.nat.uni-magdeburg.de/packing/cci/
I guess we don't need more than 6 significant figures realistically?

It also provides near optimal paths spanning the instances, "All optimal TSP tours", but they're distributed across pdfs. If there's any interest in doing this, I can email the author and ask if they can post the tours in plain text. Since the extrusion layers of all the instances - if not rotated, as they don't need to be - start and stop at the same place, that will provide a good printing order. Obviously the path should be followed forward for one layer, then reverse, then forward again and so forth.

u89djt, 2024-03-22:
@RichiH @pstrobl96 I kept going and published a couple of bed layouts: https://www.printables.com/model/815386-xl-bed-disccoin-packings-83x40mmclearance-and-35x6
I didn't include your specific object in that post, so here it is: 
[83-grafana-coins-on-XL-bed.zip](https://github.com/prusa3d/PrusaSlicer/files/14727048/83-grafana-coins-on-XL-bed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]