prusa3d/PrusaSlicer #12466: Crash on 2.7.2 on Debian

Tyler-Hartwig-Pariveda, 2024-03-14:
### Description of the bug

I did a fresh install of PrusaSlicer 2.7.2 on a Debian 11 instance. Within I used the configuration wizard to setup a new printer (Snapmaker 2.0 Dual Extrusion). A added a Snapmaker PLA Filament and clicked Finish. PrusaSlicer crashed at this point.

### Project file & How to reproduce

1. Install PrusaSlicer fresh on Debian 11
2. Open PrusaSlicer and start configuration wizard
3. Configure a Snapmaker 2.0 With Dual Extrusion
4. Add Snapmaker PLA as a filament 
5. Continue though wizard and click finish
6. PrusaSlicer should crash

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2-Linux (Intel 64-bit GTK3)

### Operating system

Debian 11

### Printer model

Snapmaker 2.0

Jan-Soustruznik, 2024-03-15:
Hello, @Tyler-Hartwig-Pariveda , than you for your report, but we cannot reproduce according to provided steps. We internally found some similar crash with loading profiles at Linux. We will include fix to the next alpha release. Could you check after release with new alpha, please?

tylerhartwig, 2024-03-15:
Thanks! Sorry I used my wrong account to post this, do you know when the next Alpha will be released?

Jan-Soustruznik, 2024-03-15:
soon

Jan-Soustruznik, 2024-03-21:
@tylerhartwig https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-alpha1

tylerhartwig, 2024-03-23:
@Jan-Soustruznik just tried it! sadly still crashed :( Are there helpful logs for me to post?

tylerhartwig, 2024-03-25:
I tried it again just now it and seemed to have worked, maybe it was a specific setting I had chosen before?

Jan-Soustruznik, 2024-03-26:
Hello, @tylerhartwig, Could you try check the latest beta1, released yesterday. There is one more crash fix of the pressets that was on Linux platform. This may help(cross finger). Thank you, have a nice day 

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-beta1

closed 2026-09-29 by prusaslicer-bot[bot]