prusa3d/PrusaSlicer #12467: Cooling Fan for Overhangs Fails to Turn Off for Rest of Model

Artemonim, 2024-03-15:
### Description of the bug

The Prusa slicer generated cooling for overhangs but didn't include any fan off commands in the gcode, while the rest of the model has the cooling fan completely off. The slicer's graphical interface indicates the fan should stop, as seen in the cooling percentage heatmap below.
[image]

### Project file & How to reproduce

1. Slice a model with Prusa, including areas with overhangs
2. Check the gcode for fan control commands: `M106 S0` commands after overhangs are missing.
3. The graphical interface shows the fan off in areas, conflicting with the gcode.
4. This inconsistency can lead to printing issues, as the fan remains on when it should be off.

[Fan_Off_Commands_Missing_for_Overhangs.zip](https://github.com/prusa3d/PrusaSlicer/files/14611122/Fan_Off_Commands_Missing_for_Overhangs.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Win11

### Printer model

Elegoo Neptune 3 Max

Manabarb, 2024-03-15:
It does have M107, Fan Off: https://marlinfw.org/docs/gcode/M107.html

Artemonim, 2024-03-15:
> It does have M107, Fan Off: https://marlinfw.org/docs/gcode/M107.html

I didn't know about this command. I've checked it now - the printer is running it normally.
But why did my fans keep spinning all the time... 🤔

closed 2024-03-15 by Artemonim