prusa3d/PrusaSlicer #12468: Add contrasting object edgelines/highlights/shadows to black objects

TheOneTrueRazor, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
Objects that are printed with a black material and are not selected in the editor preview, appear as a black blob in PrusaSlicer. Objects printed in any other color appear fine because all "shadows" and edge lines are drawn in black in the preview.

**Describe the solution you'd like**
Any item that is printed in black (or other dark materials) should have contrasting (white or grey perhaps?) edge lines and shadows. This way the item does not appear as just a black blob. This behavior carries over to the print preview on the XL display.

**Describe how it would work**
Define a range of dark colors and if the printed object falls within that range then use white/grey/etc. colors for edge lines and shadows.

**Describe alternatives you've considered**
N/A

**Additional context**
Please see attached images.
[image]
[image]
[image]

Tupson444, 2026-09-18:
still relevant

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16122.

closed 2026-10-21 by Jan-Soustruznik