prusa3d/PrusaSlicer #12469: Crash while slicing on Windows

MethylBromide, 2024-03-16:
### Description of the bug

Attempting to import and then slice attached STL file results in crash with no error message. Attempted to load the STL file into a new project and received a generic "import failure" error. Tried again, imported successfully, but again crashed on slicing.

### Project file & How to reproduce

[hingebug.zip](https://github.com/prusa3d/PrusaSlicer/files/14622266/hingebug.zip)
Open the project file, import the STL file. This might fail. If it fails, note the error message -- that is problem 1.
Try importing again until import succeeds. Now slice the model. This results in a crash. Sometimes the slicer view displays briefly before the crash.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11 64-bit

### Printer model

Sovol SV06+

Jan-Soustruznik, 2024-03-18:
Hello, @MethylBromide , Thank you for your report. We will try to fix in some next release. Thank you, have a nice day 

issue created: SPE-2195

closed 2024-06-21 by lukasmatena

lukasmatena, 2024-06-21:
Should be fixed in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Closing.

commit 5f843cc9 (2024-06-21): Fix #12469 - crash caused by bridge expansion.

Handle a very rare case in which the algorithm picks a wrong
polygon for expansion seed.