prusa3d/PrusaSlicer #12471: Hidden objects still shown on print platter in 3D view

Sharphrp, 2024-03-16:
### Description of the bug

When a print plate is loaded up with several objects, each with multiple instances then some of these are turned off (made invisible by toggling the "eye" icon), they are still shown on the 3D print platter.
Previous versions of the software didn't have this behaviour as it showed them in much darker/greyed-out shading so it was obvious which objects are going to print, and which are going to be omitted.

Can this functionality be checked, and reinstated if possible - as its very difficult to do a small test print run, or only select one or two if remaining print filament is limited.

Attached are two screenshots showing the 3D view and the Sliced view - the "eye" icons have organge lines across those instances being hidden, but in the 3D view its not possible to identify which these are on the platter as everything is shaded the same.

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3S + MMU2

neophyl, 2024-03-17:
That hasn't been changed as far as I'm aware.  It does still show the non printing ones in a different shade. 
[image]

I'd suggest it is your choice of filament rendering colours that is causing the issue.  They all appear to be different shades of grey.

Jan-Soustruznik, 2024-03-18:
Hello, @Sharphrp , could you share your 3MF(project) file, please? Thank you

closed 2026-09-29 by prusaslicer-bot[bot]