prusa3d/PrusaSlicer #12474: Slicer hangs on "Preparing infill. 30%." when slicing larger files at 5% gyroid infill.

Enavia, 2024-03-18:
### Description of the bug

Slicer hangs on "Preparing infill. 30%." when slicing larger files at 5% gyroid infill. Does not make further progress after multiple hours.

### Project file & How to reproduce

[Wall with Towers.zip](https://github.com/prusa3d/PrusaSlicer/files/14629535/Wall.with.Towers.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Windows 11 home

### Printer model

XL

Jan-Soustruznik, 2024-03-21:
Hello, @Enavia , sorry but we cannot reproduce your reported issue. We tried it on some different PCs with different HW, but the prvided 3MF was sliced without problems. Could you check if you shared correct 3MF(settings). BTW this problem starts in 2.7.2 or have you it also at the older versions of prusaSlicer?

bunnybones1, 2026-09-25:
I had this issue today, and while I did find a strange workaround (rotate your model even 0.1 degrees on the plate), it is definitely a bug, and [here is the fix](https://github.com/prusa3d/PrusaSlicer/pull/15900)