prusa3d/PrusaSlicer #12476: When the G-code is set to Klipper, the M601 should be changed for pause.

SphaeroX, 2024-03-18:
### Description of the bug

Currently, when adding a pause at a specific layer height in the preview, M601 is always added to the G-code. However, M601 does not exist in Klipper, so PrusaSlicer should (even afterwards, from my point of view) change this M601 command to PAUSE.

### Project file & How to reproduce

not needed

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Win11

### Printer model

custom

SphaeroX, 2024-03-18:
> I disagree. The codes in the printer configuration for those are all configurable already and have no relation to the firmware type you select. Its easy to modify those for your printer. If the printer configuration is misconfigured then thats a different issue.
> 
> Also as Klipper is pretty much configurable to however the user wants it is very easy to make the Pause work on any Klipper setup with whatever code you want to use.
> 
> Forcing the use of a specific code when the output type is 'Klipper' is contrary to that configurability.

I know one can fiddle around with macros again, but if a slicer tells me it's generating the code for this firmware, then in my opinion, all commands should work for this firmware as well.

rtyr, 2024-03-18:
> I know one can fiddle around with gocde replacements.

No need for that, Pause g-code is literally configurable in the printer settings, so you can set whatever g-code you want.

[image]

closed 2024-03-18 by rtyr