prusa3d/PrusaSlicer #12477: Prusaslicer 2.7.2 Linux saving Gcode to SD card

Longwilly, 2024-03-18:
### Description of the bug

When saved to SD card, Mk3S printer shows file as incomplete. When I save to hard drive and then transfer to SD card file works normally and I can print it. Hence SD card is good. One other thing to note is that file exported directly to SD card shows up with appropriate size e.g 189kB. After trying in printer (and shows as incomplete) and then put back into laptop and examined it shows 0kB.

### Project file & How to reproduce

[Large box tray corner support-Body.3mf.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/14639894/Large.box.tray.corner.support-Body.3mf.tar.gz)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Linux Mint Debian Edition Faye

### Printer model

Ptusa MK3S+

starthal, 2024-03-22:
Are you ejecting the SD card before removing it from the PC, either from the button in PS or from the file manager?

https://help.prusa3d.com/article/export-sd-usb-flash-drive_120549

Longwilly, 2024-03-22:
Thank you for the input and the link. I've never used the eject feature before, but I've only recently switched to Linux. I will give that a try. Thank you.

starthal, 2024-03-22:
MacOS and Windows will usually give you an alert if you remove a drive without ejecting. That's not always the case with GNU/Linux, depending on the distribution.

An OS may buffer write operations to a drive -- that is, keep the data in RAM and not perform the physical write until it's more appropriate to do so, or until forced to do so. The "eject" operation forces all pending writes to complete so that the drive can be removed.

Longwilly, 2024-03-22:
Thank you again. I'll be trying it out this weekend.
Bill M

hm2dev, 2024-03-22:
I did get a similar error after updating to Slicer 2.7.2 in the last few days.
I am using the Windows version.

This might be a different problem, but I first also thought it is related to SD card writing.

The printer would spit out "file incomplete", "sliced for different printer" or similar error messages after loading the file, but not reproducibly so.

After updating the configuration files today the printer then started complaining about a new firmware being available, while the above messages stayed the same.
These messages then disappeared after updating printer firmware to 3.13.3 and then MMU Firmware to 3.0.2.

There are still some other issues with MMU load, so currently I can't print with MMU with filed generated from 2.7.2.
The printer loads to the extruder and then back to MMU and then repeats the process several times until finally giving up with https://prusa.io/04108. Manually loading filament into the extruder (e.g. via heating up and moving the extruder via the menu) loads and extrudes fine. The sensors also work correctly and the filament tip is good. The problem started after updating to 2.7.2.

I will try PrusaSlicer 2.7.3-alpha1 or try to go back to 2.7.1 although I am not sure that will work/solve the problem, now the firmware is updated.

closed 2026-09-29 by prusaslicer-bot[bot]