prusa3d/PrusaSlicer #12479: Prusa Slicer crashes on complex models everytime, regardless of support type or size.  Full program close and loss of all model info and settings changes

3dDesigner360, 2024-03-18:
### Description of the bug

The problem first occurred when I upgraded my PC and added a new processor, mob, and RAM. This does not happen when using Cura. I hate having to bounce between the two for the same project. This did not occur on my older system no matter the complexity or support style. This was also 2 to 3 updates back at this point, being the last update landed today for me.  I can run very small complex shapes but as soon as it reaches softball size it's a gamble.  I have run several stress tests on the processor RAM and mobo just to make sure it's not a hardware issue and even sent back my processor and it is still occurring. This is impacting my work and business greatly and has caused hours of delays in testing. ill list hardware specs and operating system below for reference just in case there is an incompatability issue. 

Windows 10 
13th gen intel core  i9 - 13900k
gskill 16 gig ram x4 
NVIDIA GeForceGTX 1080 
Samsung SSD 980 pro and Adata sx850  

### Project file & How to reproduce

[toe cap .zip](https://github.com/prusa3d/PrusaSlicer/files/14643083/toe.cap.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

prusa slicer 2.7.2

### Operating system

windows 10 

### Printer model

MK3s +

jb221, 2024-03-20:
Do you have XMP enabled in BIOS? I had a very similar issue that popped up all of a sudden one day, after testing everything (including removing ram sticks in pairs and trying different slot configurations) everything tested good. My friend suggested turning off XMP, I did it to prove it wouldn't be that and lo and behold the issue vanished. When this was happening PrusaSlicer was the only program giving me any issues and everything else tested good.

It's probably a long shot that it's the same problem you're having, but easy enough that it's worth testing.

3dDesigner360, 2024-03-21:
ok update 3/21/24 
changed out the processor and everything was great for about a day. sliced any 3d model I wanted and all was good. 
Until this morning, I'm back to it crashing on just about every model I try, as suggested I went into the bios and turned off XMP and this changed nothing. Thank you for the suggestion though.  I'm so frustrated at this point I have no idea what to do. It literally worked yesterday.

starthal, 2024-03-22:
What stress tests did you run on RAM? I'm not an expert but I suspect some of these are more focused on overall stability than individual bit errors.
Consider [Memtest86+](https://www.memtest.org/) to ensure you are checking all of your RAM.

3dDesigner360, 2024-03-25:
did that prior and again to just test it but both the new and old ram, ( identical sets of skill ) passed perfectly. 

this is another observation I have noticed it the processor is taxed with multiple programs running and a processor stress test going at the same time it slices fine. repeated this multiple time to test for a fluk but this seems to be the case. which only complicated my variables and added to my frustration.

murk-sy, 2024-09-23:
I know this is an inactive issue, but I found it while searching for duplicates.

Seems the issue here was a now fairly well known hardware issue with Intel processors
https://www.tomshardware.com/pc-components/cpus/is-your-intel-core-i9-13900k-crashing-in-games-your-motherboard-bios-settings-may-be-to-blame-other-high-end-intel-cpus-also-affected
(Relatively old article, new information came to light since, your CPU might need a warranty replacement)

closed 2026-09-29 by prusaslicer-bot[bot]