prusa3d/PrusaSlicer #12480: "Slow down if layer print time is below:" but only for certain parts of the print.

cray12399, 2024-03-18:
**Is your feature request related to a problem? Please describe.**
I am currently tuning my printer for speed and cooling is a major bottleneck to my ability to print quality prints at a fast speed. I like the feature to slow down a print if the layer time is short. However, I don't like that it slows the entire print rather than just key areas where the quality is severely affected by bad cooling. I would like to only slow down print speed on visible sections like outer perimeters only. Sections like infill dont matter because they are not visible anyways.

**Describe the solution you'd like**
Just add an advanced/expert option to only slow down the print speed for certain parts of a print. I don't imagine it would even be that difficult of a feature to implement.

**Describe how it would work**
You could implement a category of checkboxes for each speed that is shown in the speed panel and allow users to uncheck them if they do not wish for them to be slowed down.

**Describe alternatives you've considered**
I have tried just manually going in and lowering the speed on small prints where this feature would be helpful. However, I like the automatic slow down because once I get it tuned, it really does take the guesswork out of determining what speeds my thermal performance will bottleneck me at.

**Additional context**
Add any other context or paint drawing about the feature request here.

BowlesCR, 2024-03-19:
Have you seen the Max Volumetric Speed setting? https://help.prusa3d.com/article/max-volumetric-speed_127176

Not quite what you're describing, but it might help.

cray12399, 2024-03-20:
> Have you seen the Max Volumetric Speed setting? https://help.prusa3d.com/article/max-volumetric-speed_127176
> 
> Not quite what you're describing, but it might help.

Hi thank you for the response!

Im familiar, but it's not really what I'm looking for. Basically, the "slow down if layer time is below:" works, but it slows down the whole print. I would like to be able to choose which parts of the print it slows such as external perimeters while not slowing down parts like infill.

Does that make it clear? If not, I might be able to make a graphic of what I am talking about.

vit-zikmund, 2024-04-16:
I must say this "slow down if layer time is below" feature bit me hard when trying to print some calibration cubes and stuff (not knowing about it first, wondering why the heck some layers get printed faster than others). The way how it works now makes certain "features" print at various speeds throughout the height of the object, which usually has a very visible impact on the external perimeter.
I suppose the potential slowdown for certain featured should be applied through the height of the object, or at least considering the external perimeter not to slow down the whole print for a couple fast layers.

vit-zikmund, 2026-09-21:
This is a little too eager solution - dumping legacy issues before there's a stable release... But if there's anyone willing to verify the "bug" has been tackled in the new alpha release, please, do so 🙏 Thanks!