prusa3d/PrusaSlicer #12481: optimize printing cubes PrusaSlicer

codefhacker, 2024-03-19:
**Is your feature request related to a problem? Please describe.**
When priting many small cubes for a LED diffuser Prusaslicer calculates that i must extrude around all cubes. **see the Video** this takes many time and could be done better. and more efficient

**Describe the solution you'd like**
Print first Up and then Down this is way faster see my picture

**Describe how it would work**
Maybe add a feature for it to enable.

**Describe alternatives you've considered**
None

**Additional context**
My video and the picture of the idea
[image]

https://github.com/prusa3d/PrusaSlicer/assets/65905014/df80a6a6-5b1c-40e9-930b-454f7678b1a4

codefhacker, 2024-03-19:
See step file below :
[step_file.zip](https://github.com/prusa3d/PrusaSlicer/files/14649382/step_file.zip)

u89djt, 2024-03-19:
Until the slicer can create an effective version of that for you, here's my suggestion for getting a rectangular grid of cavities to slice as a single perimeter per layer: https://www.printables.com/model/811677-workaround-for-github-request-for-more-efficient-s
I've included a 3mf file of your specific case, which includes the specification of a gap closing criterion smaller than the gaps used to divert the slicer in the model, but non-zero because the model arrives in chunks. More info in the printables post.
(not affiliated with Prusa)

codefhacker, 2024-03-20:
@u89djt  Thank you for posting I will try that out until the problem is fixed in Prusaslicer.

u89djt, 2024-03-20:
Honestly I think it isn't a problem with the slicer. A desirable feature maybe, but see image below. To get that kind of regular striping set up for a model presented as a solid lump would need tricky global analysis by the slicer, and the range of special cases and gotchas make it a hefty task to create reliable code. 
The perimeters would have to look something like this:
[image]
You can see that there are about four stops and starts with travel between them per intersection, which I think is more than you can achieve by analysing the pattern yourself to get something like the pattern I'm suggesting in the printables post. You might wonder about reducing the flow to a half as you cross the corner, but that's not trivial either. We'd need someone on the slicer team to comment on that.

codefhacker, 2024-03-20:
Yes, the problem is now known and there is a workaround, but it would still be a very nice option to add it in Prusaslicer. 

At least I have now managed to reduce the movement by your Fusion 360 idea, thanks for that

u89djt, 2024-03-20:
You're welcome, it was interesting working it through. The fundamental idea of shepherding the slicer with carefully chosen gaps and divisions of the bodies for vase mode printing to create more efficient prints is old. I developed a specific regime (well, three in total while I worked out what some of the challenges actually are, with the first two less efficient that the one I posted) for regular rectangular arrays, and I'm probably not the only person to have done essentially that. The coders definitely know that uninterrupted extrusions are faster.

Tupson444, 2026-09-18:
still relevant