prusa3d/PrusaSlicer #12482: Multimaterial Painting issue with "depth" of painting

mszeemann, 2024-03-19:
### Description of the bug

I explained the issue in detail in a post in the prusa forum. Including more pictures and project file.
Another user found a workaround but I think there is still an issue with the new voronoi diagrams. Post can be found here: https://forum.prusa3d.com/forum/original-prusa-xl-tool-changer-how-do-i-print-this-printing-help/multimaterial-painting-issue-with-depth-of-painting/

### Project file & How to reproduce

[image]
[bac_t4_open_and_close_genoma_coloured.zip](https://github.com/prusa3d/PrusaSlicer/files/14649506/bac_t4_open_and_close_genoma_coloured.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2 stable

### Operating system

Win11

### Printer model

XL5T Input shaping

Jan-Soustruznik, 2024-03-27:
Hello, @mszeemann , Thank you for your report. I think that is similar case of the multi material painting limitation of current implementation [#12495](https://github.com/prusa3d/PrusaSlicer/issues/12495) . It seams that the wrong painted perimeters are wrong decided as "inside" so is used default set extruder (extruder1; green color). I suggest to change the default extruder and modify painting the legs.

mszeemann, 2024-03-27:
Thanks for the reply!
Somebody from the Forums already suggested that and also showed, that it works. I now have the tools to work around it. So all fine. ✌🏼
But maybe sometime it might get fixed. 😊

Jan-Soustruznik, 2024-03-28:
@mszeemann , I will close this issue as duplicate. If you will have some ideas or question continue in the open issue please. We have it on your TODO list, but low priority right now.

closed 2024-03-28 by Jan-Soustruznik